sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library #6: Heading isn't working

csmith105, 2017-04-27:
Using https://www.sparkfun.com/products/13762

The compass heading is useless.

MichaelPriven, 2017-06-05:
It useless due to 2 reasons
1. the magnetometer isn't calibrated
2. there is an error in the function

> float MPU9250_DMP::computeCompassHeading(void)
{
	if (my == 0)
		**heading = (mx < 0) ? 180.0 : 0;**
	else
		heading = atan2(mx, my);
	
	if (heading > PI) heading -= (2 * PI);
	else if (heading < -PI) heading += (2 * PI);
	else if (heading < 0) heading += 2 * PI;
	
	heading*= 180.0 / PI;
	
	return heading;

> }
should be - 
heading = (mx<0) ? **PI** : 0;

csmith105, 2017-06-05:
Is this pushed?

MichaelPriven, 2017-06-05:
I'm not familiar with github, what do you mean by "pushed"?

monstah, 2017-06-14:
In general it just means "to upload the new code" (pushing)

MichaelPriven, 2017-06-14:
OK, I dont know how to push :)

monstah, 2017-06-15:
Correct me if I am wrong, as far as I know, you are not the owner of this repository, this means you either have to make a pull request in git (you will find a lot on google for this topic) and the owner has to make the changes available for everyone or.. do a fork of this project and make the changes available on your own git-repository. I think, thats basically it. 💃

commit 2b6f5129 (2018-06-25): Degrees, Radians... whatever it takes! Fixes #6 - Thx MichaelPriven
commit 8ce9837b (2018-06-25): Merge pull request #8 from bhagman/master

Degrees, Radians... whatever it takes! Fixes #6 - Thx MichaelPriven
closed 2018-06-25 by jimblom