sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library #11: Doesn't work on Arduino Uno

april4s008, 2017-08-10:
Hi thanks for the great work. But I found the sketch too large for my Arduino Uno R3 and the memory required is 188% which results in an error.
Is there any MPU 9250 DMP library that works for Uno?

Many thanks!

monstah, 2017-08-10:
Don't know in there is another lib for DMP stuff... another question, what data do u need/expect from it?

april4s008, 2017-08-10:
Just need yaw/pitch/roll output directly out of DMP. I've been using raw data of accelerometer/gyro/magnetometer to calculate the yaw/pitch/roll. So I just want to make it simpler using DMP and see if the output is good enough

monstah, 2017-08-10:
According to kris winer see [here](https://github.com/kriswiner/MPU6050/wiki/Affordable-9-DoF-Sensor-Fusion) and several issues in his MPU9250 [repository](https://github.com/kriswiner/MPU9250/issues) its not possible to get a reliable yaw from the DMP - the code this sketch is using is also a very old version of kris' sketches, as he said. Did you got reliable yaw/pitch/roll values with the computations from the raw values? Which algorithm did you use?

april4s008, 2017-08-10:
I actually used Kris sketch exactly. But probably newer version. With MPU 9250 yaw is pretty stable.  
I don't care too much about the yaw, as long as DMP output stable pitch and roll it would be great. The reason I want to try DMP without using other algorithm is that it's simpler.

DiegoSynth, 2017-12-29:
Hello,
I'm having exactly the same problem. I was previously using MPU 6050 with jrowberg's library (working perfectly) but I've just switched to MPU 9250 (to get a stable yaw) and the only library (so far) that worked is the SparkFun (basic example). Quaternion example (yaw / pitch / roll) won't fit in my nano (memory).

I haven't worked with raw data before, and you mentioned you managed to calculate YPR with it. Do you think you could probably share the code with me (as I'm really not very good in maths) so I skip this pain? Thank you very much!

JohnCalin, 2018-12-21:
He guys, I had the same issue with my Nano, and found a solution.  You can modify the library to store the gigantic firmware image in flash instead of dynamic memory.
Patch:

```
diff -Naur SparkFun_MPU-9250-DMP_Arduino_Library-master/src/util/inv_mpu.c SparkFun_MPU-9250-DMP_Arduino_Library-mod/src/util/inv_mpu.c
--- SparkFun_MPU-9250-DMP_Arduino_Library-master/src/util/inv_mpu.c     2018-12-21 10:53:53.474317242 -0800
+++ SparkFun_MPU-9250-DMP_Arduino_Library-mod/src/util/inv_mpu.c        2018-12-21 10:29:54.406243820 -0800
@@ -2780,7 +2780,7 @@
     unsigned short this_write;
     /* Must divide evenly into st.hw->bank_size to avoid bank crossings. */
 #define LOAD_CHUNK  (16)
-    unsigned char cur[LOAD_CHUNK], tmp[2];
+    unsigned char cur[LOAD_CHUNK], fwbuff[LOAD_CHUNK], tmp[2];
 
     if (st.chip_cfg.dmp_loaded)
         /* DMP should only be loaded once. */
@@ -2790,11 +2790,13 @@
         return -1;
     for (ii = 0; ii < length; ii += this_write) {
         this_write = min(LOAD_CHUNK, length - ii);
-        if (mpu_write_mem(ii, this_write, (unsigned char*)&firmware[ii]))
+        for(int x =0;x<this_write;x++)
+          fwbuff[x] = pgm_read_word_near(firmware+ii+x);
+        if (mpu_write_mem(ii, this_write, fwbuff))
             return -1;
         if (mpu_read_mem(ii, this_write, cur))
             return -1;
-        if (memcmp(firmware+ii, cur, this_write))
+        if (memcmp(fwbuff, cur, this_write))
             return -2;
     }
 
diff -Naur SparkFun_MPU-9250-DMP_Arduino_Library-master/src/util/inv_mpu_dmp_motion_driver.c SparkFun_MPU-9250-DMP_Arduino_Library-mod/src/util/inv_mpu_dmp_motion_driver.c
--- SparkFun_MPU-9250-DMP_Arduino_Library-master/src/util/inv_mpu_dmp_motion_driver.c   2018-12-21 10:53:53.474317242 -0800
+++ SparkFun_MPU-9250-DMP_Arduino_Library-mod/src/util/inv_mpu_dmp_motion_driver.c      2018-12-21 10:07:37.194074031 -0800
@@ -220,7 +220,7 @@
 
 #define DMP_CODE_SIZE           (3062)
 
-static const unsigned char dmp_memory[DMP_CODE_SIZE] = {
+static const unsigned char dmp_memory[DMP_CODE_SIZE] PROGMEM = {
     /* bank # 0 */
     0x00, 0x00, 0x70, 0x00, 0x00, 0x00, 0x00, 0x24, 0x00, 0x00, 0x00, 0x02, 0x00, 0x03, 0x00, 0x00,
     0x00, 0x65, 0x00, 0x54, 0xff, 0xef, 0x00, 0x00, 0xfa, 0x80, 0x00, 0x0b, 0x12, 0x82, 0x00, 0x01,

```

ahckmed, 2019-02-21:
HERE -> https://github.com/kharlashkin/MPU-9250DMP

santaimpersonator, 2019-03-15:
I am closing this issue because this library is only tested and supported for SAMD microprocessors; specifically the SparkFun Razor IMU (see [README.md](https://github.com/sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library/blob/master/README.md)).

- For assistance with finding another DMP, please use our [forum](https://forum.sparkfun.com/) to reach out to the SparkFun community.
- For further assistance with the yaw accuracy, please file/use a separate issue or use our [forum](https://forum.sparkfun.com/) for technical assistance.

santaimpersonator, 2019-03-15:
- For assistance with finding another DMP, please use our [forum](https://forum.sparkfun.com/) to reach out to our technical support team and the SparkFun community.- - For further assistance with the yaw accuracy, please file a separate issue or use our [forum](https://forum.sparkfun.com/) for technical as

closed 2019-03-15 by santaimpersonator