sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library #18: Custom Sercom Ports

dminskiy, 2018-02-19:
Hello!
I've been trying to change standard Wire to the custom pins 3 and 4. 
I tried my best which is definitely not enough. I've done it with other sensors, they worked fine. It's just me not being able to put the code in the right place. 
Can you give any advice on that, please?

Here's the code I'm trying out:

```
#include <Wire.h>
#include "wiring_private.h" // pinPeripheral() function
TwoWire IMUWire(&sercom2, 4, 3);
    // Assign pins 4 & 3 to SERCOM functionality
    pinPeripheral(4, PIO_SERCOM_ALT);
    pinPeripheral(3, PIO_SERCOM_ALT);
    IMUWire.begin();
```

P.S. I tried the beginning of .begin() func in ...MPU9250-DMP.cpp, inv_mpu.c and arduino_mpu9250_i2c.c

Thank you!
Dmitry

oclyke, 2018-08-03:
So I spent a while longer trying to add flexibility to what I2C port you use. The way I see it there are two parts to the challenge:

1) The DMP version of the library is based on a set of functions provided by InvenSense (you can find them in the src/util folder). These functions are pretty rigid - they have hardcoded data structures that contain device-specific values such as the I2C address (defaulting to 0x68). They specified an interface that requires I2C. I think that this part will be easy to modify to accept other valid I2C ports.

2) The other challenge is to reconfigure the peripherals and multiplexing so that your new pins can actually use Wire. From what I can tell (I'm new to the SAMD21) you will effectively be making a new "variant" of the board that includes more I2C peripherals and fewer other serial interfaces. 

In the future I think it would be cool to support an easy way to reconfigure the board into custom variants. Please don't hesitate to shoot any more questions this way!

bboyho, 2019-03-13:
Hi @dminskiy ,

I'm not sure if you got this working already but you should check out this neat tutorial that might be of some use on adding additional SERCOM ports => [ [https://learn.sparkfun.com/tutorials/adding-more-sercom-ports-for-samd-boards/all#adding-an-i2c](https://learn.sparkfun.com/tutorials/adding-more-sercom-ports-for-samd-boards/all#adding-an-i2c).  I haven't tried this specifically with the 9DoF Razor. However, I was able to add custom SERCOM ports on the SAMD21 RedBoard Turbo.  You'll have to check the board defs and some datasheets for the 9DoF Razor to see if it is possible.

closed 2019-03-13 by bboyho