sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library #22: Quaternion DMP calulation not working on Mega 2560

raomin, 2018-07-31:
I ran MPU9250_DMP_Quaternion.ino without any problem on an ESP8266.
Now I want to use it on a Arduino Mega 2560 and it returns the following:

```
Q:    INF,    INF,    INF,    INF
R/P/Y:  NAN, 180.00,  NAN
Time: 32761 ms
```

MPU9250_DMP_Gyro_Cal.ino works correctly on the same Arduino Mega 2560.

I tried DMP_FEATURE_6X_LP_QUAT and DMP_FEATURE_LP_QUAT both gives the same.

Also, dmpBegin returns INV_SUCCESS.

I also tried with Wire.setClock(400000), no success...

Any idea what could cause this issue?

kolosy, 2018-08-01:
I'm actually running into the same exact issue (same mcu, too). Still debugging, but my hunch is that it's the function that converts the quaternion values from integer math to floats. The raw values seem legit, but go to INF after the call to qToFloat()

kolosy, 2018-08-01:
Yeah, confirmed. I'll try to get a PR open when I have a minute, but edit `MPU9250_DMP::qToFloat` and change the integer literals from `1` and `2` to `1L` and `2L`, respectively. That got it working for me.

kolosy, 2018-08-09:
I don't see that on mine.. not sure how it's different from yours

pradeep14581, 2018-08-10:
Hello everyone. I am also facing the same issue. DId anyone get the solution

Papierkor654, 2019-05-13:
> Yeah, confirmed. I'll try to get a PR open when I have a minute, but edit `MPU9250_DMP::qToFloat` and change the integer literals from `1` and `2` to `1L` and `2L`, respectively. That got it working for me.

Hey kolosy
Just wanted to say **thank you**. I've been looking for a way to read this IMU for about 1 month now. With your solution everything works. 
Thank you so much and have a great day ^^

raomin, 2019-05-14:
I'll close this issue as @kolosy found a solution for the NAN.

kolosy, 2019-05-14:
@raomin I'd recommend keeping it open until it's actually fixed in the codebase...

raomin, 2019-05-14:
good point.