sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library #30: Bricked 3 mkr zero boards with this card - advice please!! :(

jon2000, 2019-05-10:
I am trying to get the 9250 working but am falling at the first hurdle.  I am trying to run the code at the hookup guide tutorial, but after it compiles and runs my mkrzero is then bricked.  By board three I was super careful - checking resistance between vcc/gnd on the 9250 to check for shorts and I could hook up the board to the mkr zero vcc,gnd, scl and sda and run 'blink' and all is fine - nothing blows yet.  BUT as soon as I run the example code, that's it - nothing.  
  Any advice very gratefully received!!

bboyho, 2019-05-16:
Hi,

I have not used the Arduino Zero on the MPU9250 before but it is a SAMD21. Try hitting the reset button twice to see if it re-enumerates if you have not already. The microcontroller has a built-in CDC so there are two COM ports as stated in the [MKRZero's FAQ](https://store.arduino.cc/usa/arduino/arduino-mkr-family/arduino-mkrzero). From my experience with the SAMD21 with a different project, it may switch to another COM port after uploading code. 

Looking into the library, I think I might have another idea but I would need to look into it further. 

Otherwise, I recommend checking in the SparkFun Forums if you have not already. => [https://forum.sparkfun.com/](https://forum.sparkfun.com/).

jon2000, 2019-05-16:
Hey bboyho, awesome the double reset brings the boards back to life!  I thought I had read a lot but missed this trick.

closed 2019-05-16 by jon2000