sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library #32: endless loop with vibration

Papierkor654, 2019-05-31:
I'm currently building an arduino drone and I have the problem, that the IMU code gets stuck somewhere if the Propeller speed is at a certain rate. This only happens, if the IMU is sticked to the drone. If I take the IMU of the drone and hold it in my hands, there is no error and everything runs fine. 

Dos anyone have an idea why this could happen? Is it possible that the IMU gets stuck, when it experiences to much vibration?

The soldering of the IMU should be fine. There is no problem, if you touch / bend the cable.

dejanmeznarc, 2019-08-07:
Does your battery/PSU provide enough current?

Papierkor654, 2019-08-08:
Yes it should... 
But after a few days of leaving the drone alone, it just worked. Don't know why xD
But thanks for your help ^^

closed 2019-10-02 by Papierkor654