sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library #33: How to change  axis in DMP?

Sketh, 2019-08-27:
In my case i want to let 9250 zaxis pointing up, i want change 9250 axis, how can i do this?

remi0s, 2019-10-14:
any information how to change the axis? I need the z axis to become the y and the opposite.

gabriel-shanahan, 2019-10-15:
Why can't you just flip the values yourself? By "forward" I'm assuming you mean the direction the Y axis is by default. Assuming you only want to flip the Y and Z axis, you can just do the equivalent of

new_z = old_y
new_y = old_z

in your code.

remi0s, 2019-10-16:
> Why can't you just flip the values yourself? By "forward" I'm assuming you mean the direction the Y axis is by default. Assuming you only want to flip the Y and Z axis, you can just do the equivalent of
> 
> new_z = old_y
> new_y = old_z
> 
> in your code.

i wanted to use directly the quanternions inside Unity enviroment, which has different axis possition. Anyway the reason i needed this was because the roll pitch yaw values werent allways correct. Turns out to be a bug in computeEulerAngles () function as discussed [here](https://forum.sparkfun.com/viewtopic.php?f=14&t=45516). 

Using this function solved my problem:

```
static void toEulerianAngle(float w, float x, float y, float z, double& roll, double& pitch, double& yaw)
{
  double ysqr = y * y;

  // roll (x-axis rotation)
  double t0 = +2.0 * (w * x + y * z);
  double t1 = +1.0 - 2.0 * (x * x + ysqr);
  roll = atan2(t0, t1) * (180.0 / PI);;

  // pitch (y-axis rotation)
  double t2 = +2.0 * (w * y - z * x);
  t2 = t2 > 1.0 ? 1.0 : t2;
  t2 = t2 < -1.0 ? -1.0 : t2;
  pitch = asin(t2) * (180.0 / PI);;

  // yaw (z-axis rotation)
  double t3 = +2.0 * (w * z + x * y);
  double t4 = +1.0 - 2.0 * (ysqr + z * z);  
  yaw = atan2(t3, t4) * (180.0 / PI);;
}
```

gabriel-shanahan, 2019-10-16:
Thanks for the explanation and for the code.

Maybe this is what you need?

https://stackoverflow.com/questions/16099979/can-i-switch-x-y-z-in-a-quaternion

From a cursory glance, it looks like you can do the flips in quaternion form, and then just multiply the real part by (-1)^(number_of_flips).

santaimpersonator, 2020-03-16:
Sounds like users solved own issue... closing.

closed 2020-03-16 by santaimpersonator