sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library #36: ~60 MS DELAY EVERY ~.5S

skemp32, 2019-11-23:
The IMU will have these random delays/gaps every so often where it doesn't transmit for around 60ms every .5 seconds or so: This is a zoomed in picture of an Euler angle output vs the time log:

[image]

All I am doing is checking fifoAvailable, then doing dmpUpdateFifo. There are no delays or loops in my code. The dmpbegin is initialized with a fifo rate of 200 Hz,

santaimpersonator, 2019-12-27:
I would guess that this is probably one of two things, either you are logging data and the buffer being written to an SD card during that period or it is the time it takes to dump the FIFO buffer. However, without narrowing down where in the code this is occurring it is hard to investigate.

santaimpersonator, 2020-03-16:
Closing due to lack of response.

closed 2020-03-16 by santaimpersonator