sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library #39: Unable to get raw accelerometer data after calculating quaternion

lannerxiii, 2020-01-03:
Hello, I am facing a problem getting raw accelerometer data. I need to substract the gravity from my raw accel data,so I am calculating my quaternions with DPM. My code is the following

`  float q0 = imu.calcQuat(imu.qw);
  float q1 = imu.calcQuat(imu.qx);
  float q2 = imu.calcQuat(imu.qy);
  float q3 = imu.calcQuat(imu.qz);

  Serial.println("Q: " + String(q0, 4) + ", " + String(q1, 4) + ", " + String(q2, 4) + ", " + String(q3, 4));
  float g0,g1,g2 = 0;

  g0 = 2 * (q1 * q3 - q0 * q2);
  g1 = 2 * (q0 * q1 - q2 * q3);
  g2 = q0 * q0 - q1 * q1 - q2 * q2 + q3 * q3;

  ax =  imu.calcAccel(imu.ay);
  ay =  imu.calcAccel(imu.ay);
  az =  imu.calcAccel(imu.az);

  cax =  (9.8 * ax - g0);
  cay =  (9.8 * ay - g1);
  caz =  (9.8 * az - g2);
  Serial.println("A: " + String(g0, 4) + ", " + String(g1, 4) + ", " + String(g2, 4) );`

Im using the DPM Quaternion example and made those modifications, but my ax, ay and Az values are zero.  The main loop it's exactly the one from the example, I only removed the call to computeEulerAngles() because I do not need it. I think Im not understanding exactly how to get the raw data, because the basic example is different, also I know understand that the dpm code utilizes a FIFO Buffer to store the data.  I need to get the raw AY,AY and AZ values that where used to calculate the quaternion. Am I doing something wrong? 

Thank you

cristian1706, 2020-01-03:
I am with the same problem... if anyone could help us I would appreciate it

beegee-tokyo, 2020-01-23:
I got it to work with both DMP FiFo AND mag values by initializing like this:
```cpp
	if (imu.begin() != INV_SUCCESS)
	{
		while (1)
		{
			Serial.println("Unable to communicate with MPU-9250");
			Serial.println("Check connections, and try again.");
			Serial.println();
			delay(5000);
		}
	}

	imu.dmpBegin(DMP_FEATURE_6X_LP_QUAT |		  // Enable 6-axis quat
					 DMP_FEATURE_GYRO_CAL |		  // Use gyro calibration
					 DMP_FEATURE_SEND_RAW_ACCEL | // Send raw accelerometer values to FIFO
					 DMP_FEATURE_SEND_RAW_GYRO,   // Send raw gyroscope values to FIFO
				 10);							  // Set DMP FIFO rate to 10 Hz

	// // Enable all sensors:
	imu.setSensors(INV_XYZ_GYRO | INV_XYZ_ACCEL | INV_XYZ_COMPASS);

	// Likewise, the compass (magnetometer) sample rate can be
	// set using the setCompassSampleRate() function.
	// This value can range between: 1-100Hz
	imu.setCompassSampleRate(10); // Set mag rate to 10Hz
```
and then in my loop I read the data like this
```cpp
	if (imu.dataReady())
	{
		imu.update(UPDATE_COMPASS);
	}

	// Check for new data in the FIFO
	if (imu.fifoAvailable())
	{
		// Use dmpUpdateFifo to update the ax, gx, mx, etc. values
		if (imu.dmpUpdateFifo() == INV_SUCCESS)
		{
			// computeEulerAngles can be used -- after updating the
			// quaternion values -- to estimate roll, pitch, and yaw
			imu.computeEulerAngles();
		}
	}
	Serial.printf("FLOAT:\t%.2f\t%.2f\t%.2f\t%.2f\t%.2f\t%.2f\t%.2f\t%.2f\t%.2f\n",
				  imu.calcAccel(imu.ax), imu.calcAccel(imu.ay), imu.calcAccel(imu.az),
				  imu.calcGyro(imu.gx), imu.calcGyro(imu.gy), imu.calcGyro(imu.gz),
				  imu.calcMag(imu.mx), imu.calcMag(imu.my), imu.calcMag(imu.mz));
	Serial.printf("Roll: %.2f\tPitch: %.2f\tYaw: %.2f\n",
				  imu.roll,
				  imu.pitch,
				  imu.yaw);
```

cristian1706, 2020-01-27:
> I got it to work with both DMP FiFo AND mag values by initializing like this:
> 
> ```c++
> if (imu.begin() != INV_SUCCESS)
> {
> 	while (1)
> 	{
> 		Serial.println("Unable to communicate with MPU-9250");
> 		Serial.println("Check connections, and try again.");
> 		Serial.println();
> 		delay(5000);
> 	}
> }
> 
> imu.dmpBegin(DMP_FEATURE_6X_LP_QUAT |		  // Enable 6-axis quat
> 				 DMP_FEATURE_GYRO_CAL |		  // Use gyro calibration
> 				 DMP_FEATURE_SEND_RAW_ACCEL | // Send raw accelerometer values to FIFO
> 				 DMP_FEATURE_SEND_RAW_GYRO,   // Send raw gyroscope values to FIFO
> 				 10);							  // Set DMP FIFO rate to 10 Hz
> 
> // Enable all sensors:
> imu.setSensors(INV_XYZ_GYRO | INV_XYZ_ACCEL | INV_XYZ_COMPASS);
> 
> // Likewise, the compass (magnetometer) sample rate can be
> // set using the setCompassSampleRate() function.
> // This value can range between: 1-100Hz
> imu.setCompassSampleRate(10); // Set mag rate to 10Hz
> ```
> 
> and then in my loop I read the data like this
> 
> ```c++
> if (imu.dataReady())
> {
> 	imu.update(UPDATE_COMPASS);
> }
> 
> // Check for new data in the FIFO
> if (imu.fifoAvailable())
> {
> 	// Use dmpUpdateFifo to update the ax, gx, mx, etc. values
> 	if (imu.dmpUpdateFifo() == INV_SUCCESS)
> 	{
> 		// computeEulerAngles can be used -- after updating the
> 		// quaternion values -- to estimate roll, pitch, and yaw
> 		imu.computeEulerAngles();
> 	}
> }
> Serial.printf("FLOAT:\t%.2f\t%.2f\t%.2f\t%.2f\t%.2f\t%.2f\t%.2f\t%.2f\t%.2f\n",
> 			  imu.calcAccel(imu.ax), imu.calcAccel(imu.ay), imu.calcAccel(imu.az),
> 			  imu.calcGyro(imu.gx), imu.calcGyro(imu.gy), imu.calcGyro(imu.gz),
> 			  imu.calcMag(imu.mx), imu.calcMag(imu.my), imu.calcMag(imu.mz));
> Serial.printf("Roll: %.2f\tPitch: %.2f\tYaw: %.2f\n",
> 			  imu.roll,
> 			  imu.pitch,
> 			  imu.yaw);
> ```

Thank you very much, I will be testing this soon! :D

beegee-tokyo, 2020-02-02:
Weird. My above solution worked perfectly with a nRF52832 Feather board and a ESP32 Feather board, but fails on my custom board with an Insight ISP4520 (Nordic nRF52832 + Semtech SX1262 LoRa transceiver). 
Still investigating.

cristian1706, 2020-02-03:
> Weird. My above solution worked perfectly with a nRF52832 Feather board and a ESP32 Feather board, but fails on my custom board with an Insight ISP4520 (Nordic nRF52832 + Semtech SX1262 LoRa transceiver).
> Still investigating.

Thanks! Please let me know